RainbowRobotics/rby1-sdk #8: CMake Error: Could not read presets from /rby1-sdk No "version" field

bxtbold, 2024-09-24:
안녕하세요, sdk 빌드하는 중에 이와 같은 오류가 났습니다.

cmake 버전은 이슈 #2 에 나와있는 해결방법 처럼 3.30.2로 설치되었는데 다음과 같은 오류가 납니다. 다른 해결 방법이 혹시 있을까요?
 
```bash
$ cmake --version
cmake version 3.30.2
```

```
$ cmake --preset conan-release -D BUILD_EXAMPLES=ON
CMake Error: Could not read presets from /home/user/rby1-sdk:
No "version" field
```

환경: Ubuntu 22.04, AMD64

messy-snail, 2024-09-24:
안녕하세요. 레인보우로보틱스 강한솔입니다.
[여기](https://rainbowrobotics.github.io/rby1-dev/docs/development/troubleshooting/build-issue.html)와 동일한 이슈로 생각됩니다.

[여기](https://rainbowrobotics.github.io/rby1-dev/docs/development/getting-started/installation.html)에 있는 CMake 설치 과정을 수행해보시기 바랍니다

의심되는 것은 기존 CMake가 제대로 지워지지않았을 것 같습니다.

bxtbold, 2024-09-25:
다음 명령까지 잘 실행이 되어서 이제는 이런 오류가 납니다.
cmake --build --preset conan-release

protobuf 버전이 맞지 않은 것 같은데 확인부탁드립나.

<details>
<summary>conan install . -s build_type=Release -b missing -of build</summary>

```
======== Input profiles ========
Profile host:
[settings]
arch=x86_64
build_type=Release
compiler=gcc
compiler.cppstd=gnu17
compiler.libcxx=libstdc++11
compiler.version=11
os=Linux

Profile build:
[settings]
arch=x86_64
build_type=Release
compiler=gcc
compiler.cppstd=gnu17
compiler.libcxx=libstdc++11
compiler.version=11
os=Linux

======== Computing dependency graph ========
Graph root
    conanfile.py (rby1-sdk/0.1): /home/user/rby1-sdk/conanfile.py
Requirements
    abseil/20230802.1#a8d06333152eb1926b5dbc51e3bb218d - Cache
    bzip2/1.0.8#d00dac990f08d991998d624be81a9526 - Cache
    c-ares/1.33.1#99a35827eebc984885f60362660ea232 - Cache
    eigen/3.4.0#2e192482a8acff96fe34766adca2b24c - Cache
    grpc/1.54.3#e0844c82328a52eea475a9090db20bc9 - Cache
    libcap/2.69#7ef2d60864c2e58e89db957be936dc49 - Cache
    libmount/2.39.2#64850679fa9d5ead878e5076b0cab2a8 - Cache
    libselinux/3.6#5a78ff6ae5034eeaac8da723361a8ce4 - Cache
    libsystemd/255#8e49d0c6b1d63861e3e10665cc1b7b4d - Cache
    libxcrypt/4.4.36#4b4e8f20794f1997dd59eeed0b7cdcfb - Cache
    lz4/1.9.4#652b313a0444c8b1d60d1bf9e95fb0a1 - Cache
    nlohmann_json/3.11.3#45828be26eb619a2e04ca517bb7b828d - Cache
    openssl/3.3.2#9f9f130d58e7c13e76bb8a559f0a6a8b - Cache
    pcre2/10.42#576680686b88a4dadf6d382d6d6351d9 - Cache
    protobuf/3.21.12#2099e6c66fcd31b52c3afc0c096c1d02 - Cache
    re2/20230301#e8d673155a0494f27c7589b611e8350d - Cache
    tinyxml2/10.0.0#832c7f8038cef53f6427577d6e66a7a8 - Cache
    xz_utils/5.4.5#b885d1d79c9d30cff3803f7f551dbe66 - Cache
    zlib/1.3.1#f52e03ae3d251dec704634230cd806a2 - Cache
    zstd/1.5.5#1f239731dc45147c7fc2f54bfbde73df - Cache
Build requirements
    autoconf/2.71#f9307992909d7fb3df459340f1932809 - Cache
    automake/1.16.5#058bda3e21c36c9aa8425daf3c1faf50 - Cache
    flex/2.6.4#e35bc44b3fcbcd661e0af0dc5b5b1ad4 - Cache
    gnu-config/cci.20210814#dc430d754f465e8c74463019672fb97b - Cache
    gperf/3.1#1d622ad9717e9348ed3685c9994ad0b9 - Cache
    libtool/2.4.7#08316dad5c72c541ed21e039e4cf217b - Cache
    m4/1.4.19#b38ced39a01e31fef5435bc634461fd2 - Cache
    meson/1.2.2#21b73818ba96d9eea465b310b5bbc993 - Cache
    meson/1.4.0#2262941cc8fbb0099dd0c196ca2a6c01 - Cache
    ninja/1.12.1#fd583651bf0c6a901943495d49878803 - Cache
    pkgconf/2.1.0#27f44583701117b571307cf5b5fe5605 - Cache
    pkgconf/2.2.0#6462942a22803086372db44689ba825f - Cache
    protobuf/3.21.12#2099e6c66fcd31b52c3afc0c096c1d02 - Cache
    zlib/1.3.1#f52e03ae3d251dec704634230cd806a2 - Cache
Resolved version ranges
[... 494 lines cut ...]
10690 |         1, _Internal::command_header_feedback(this), target, stream);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/user/.conan2/p/proto5794f10f7a310/p/include/google/protobuf/parse_context.h:50,
                 from /home/user/.conan2/p/proto5794f10f7a310/p/include/google/protobuf/map_type_handler.h:37,
                 from /home/user/.conan2/p/proto5794f10f7a310/p/include/google/protobuf/map.h:61,
                 from /usr/include/google/protobuf/generated_message_table_driven.h:34,
                 from /home/user/rby1-sdk/build/_protobuf/rb/api/basic_command.pb.h:26,
                 from /home/user/rby1-sdk/build/_protobuf/rb/api/basic_command.pb.cc:4:
/home/user/.conan2/p/proto5794f10f7a310/p/include/google/protobuf/wire_format_lite.h:631:19: note: candidate: ‘static uint8_t* google::protobuf::internal::WireFormatLite::InternalWriteMessage(int, const google::protobuf::MessageLite&, int, uint8_t*, google::protobuf::io::EpsCopyOutputStream*)’
  631 |   static uint8_t* InternalWriteMessage(int field_number,
      |                   ^~~~~~~~~~~~~~~~~~~~
/home/user/.conan2/p/proto5794f10f7a310/p/include/google/protobuf/wire_format_lite.h:631:19: note:   candidate expects 5 arguments, 4 provided
/home/user/rby1-sdk/build/_protobuf/rb/api/basic_command.pb.cc: In static member function ‘static const rb::api::RealTimeControlCommand& rb::api::RealTimeControlCommand::default_instance()’:
/home/user/rby1-sdk/build/_protobuf/rb/api/basic_command.pb.cc:10824:38: error: ‘InitSCC’ is not a member of ‘google::protobuf::internal’
10824 |   ::PROTOBUF_NAMESPACE_ID::internal::InitSCC(&::scc_info_RealTimeControlCommand_rb_2fapi_2fbasic_5fcommand_2eproto.base);
      |                                      ^~~~~~~
/home/user/rby1-sdk/build/_protobuf/rb/api/basic_command.pb.cc:10824:49: error: ‘::scc_info_RealTimeControlCommand_rb_2fapi_2fbasic_5fcommand_2eproto’ has not been declared; did you mean ‘InitDefaultsscc_info_RealTimeControlCommand_rb_2fapi_2fbasic_5fcommand_2eproto’?
10824 |   ::PROTOBUF_NAMESPACE_ID::internal::InitSCC(&::scc_info_RealTimeControlCommand_rb_2fapi_2fbasic_5fcommand_2eproto.base);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 InitDefaultsscc_info_RealTimeControlCommand_rb_2fapi_2fbasic_5fcommand_2eproto
gmake[2]: *** [CMakeFiles/proto-objects.dir/build.make:602: CMakeFiles/proto-objects.dir/_protobuf/rb/api/basic_command.pb.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:370: CMakeFiles/proto-objects.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2

```

messy-snail, 2024-09-25:
혹시 기존에 설치되었던 혹은 설치했던 protoc이 존재하나요?
지우고 진행해주셔야합니다. conan에서 protoc 패키지를 관리하므로 별도로 설치해줄 필요가 없습니다.

```bash
#apt로 설치한 경우
sudo apt-get remove --purge protobuf-compiler

#snap으로 설치한 경우
sudo snap remove protoc

#아래 명령어를 수행했을 때 버전이 아무것도 뜨지 않아야함.
protoc --version
```

기존의 build 폴더 삭제
아래 명령어 다시 수행

```bash
conan install . -s build_type=Release -b missing -of build
cmake --preset conan-release -D BUILD_EXAMPLES=ON
cmake --build --preset conan-release
```

---
# Question:
I ran the following commands successfully, but now I'm encountering an error when running:

```bash
cmake --build --preset conan-release
```

It seems like there is a mismatch with the protobuf version. Can you please check?

Commands I ran:
```bash
conan install . -s build_type=Release -b missing -of build
cmake --preset conan-release -D BUILD_EXAMPLES=ON
cmake --build --preset conan-release
```

```bash
  0%] Built target qdldlobject
[  4%] Built target DynamixelSDK
[ 14%] Built target OSQPLIB
[ 16%] Built target osqpstatic
[ 17%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/body_component_based_command.pb.cc.o
[ 17%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/basic_command.pb.cc.o
[ 18%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/component_based_command.pb.cc.o
[ 17%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/command_header.pb.cc.o
[ 18%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/arm_command.pb.cc.o
[ 19%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/control_manager.pb.cc.o
[ 19%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/body_command.pb.cc.o
[ 19%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/gamepad.pb.cc.o
[ 20%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/gamepad_service.pb.cc.o
[ 20%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/control_manager_service.pb.cc.o
[ 20%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/gripper_command.pb.cc.o
[ 20%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/geometry.pb.cc.o
[ 20%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/head_command.pb.cc.o
[ 21%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/gripper_command_service.pb.cc.o
[ 21%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/header.pb.cc.o
[ 22%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/log.pb.cc.o
[ 22%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/log_service.pb.cc.o
[ 22%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/mobility_command.pb.cc.o
[ 24%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/parameter.pb.cc.o
[ 24%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/ping.pb.cc.o
[ 24%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/parameter_service.pb.cc.o
[ 25%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/ping_service.pb.cc.o
[ 25%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/power.pb.cc.o
[ 27%] Built target OsqpEigen
[ 27%] Building CXX object CMakeFiles/proto-objects.dir/_protobuf/rb/api/power_service.pb.cc.o
[... 69 lines cut ...]
```

# Response:
Is it possible that there is an existing installation of `protoc` on your system? You need to remove it because `conan` manages the `protoc` package, so there's no need for a separate installation.

Run the following commands depending on how `protoc` was installed:

```bash
# If installed via apt
sudo apt-get remove --purge protobuf-compiler

# If installed via snap
sudo snap remove protoc

# After running this, `protoc --version` should not return any version.
protoc --version
```

Then, delete the existing `build` folder and re-run the following commands:

```bash
conan install . -s build_type=Release -b missing -of build
cmake --preset conan-release -D BUILD_EXAMPLES=ON
cmake --build --preset conan-release
```

bxtbold, 2024-09-26:
I ended up creating a docker container with a clean environment, but it is working.
Thank you :)

closed 2024-09-26 by bxtbold