RainbowRobotics/rby1-sdk #23: Contributing robot model to Menagerie?

kevinzakka, 2024-10-03:
As title says, it would be super cool if you would be willing to contribute your model to [MuJoCo Menagerie](https://github.com/google-deepmind/mujoco_menagerie). I would be more than happy to get it merged in + tweak things to improve the model.

messy-snail, 2024-10-22:
Thank you for your interest in our model. We are also considering contributing the model to MuJoCo Menagerie. However, we are currently focused on developing other features of the SDK and do not have the capacity at the moment. We aim to contribute the model to Menagerie in the near future.

messy-snail, 2025-01-24:
This topic is currently in progress. Please follow up at the link below:
https://github.com/google-deepmind/mujoco_menagerie/pull/134

closed 2025-01-24 by messy-snail