RainbowRobotics/rby1-sdk #29: Port mapping of rby1-sim

geniuskpj, 2024-11-08:
- 상황: 여러명이 개발하기 위해 시뮬레이터를 각자 1PC에서 여러 포트로 여러개 실행하고 싶음
- 요청:  app_main에 포트 에 대한 argument/option을 줄 수 있으면 좋겠음
- network_mode=host 대신 port mapping 사용하면 GLFW 이슈로 시뮬레이터실행 불가 
([문서](https://hub.docker.com/r/rainbowroboticsofficial/rby1-sim)내option2 실행이 가능한지 확인 바람)

messy-snail, 2024-11-08:
안녕하세요. 레인보우로보틱스 강한솔입니다.

저희 RB-Y1에 대해 귀중한 피드백을 주셔서 감사합니다.

답변 드리겠습니다.
현 구조에서는 포트를 가변적으로 변경하기가 어려운 상황입니다. 또한, `network_mode=host` 대신 port mapping을 사용해도 정상적으로 시뮬레이터가 동작합니다. 아시다시피 X11 포워딩을 위해서는 터미널에서 `xhost +` 명령어 실행이 선행되어야합니다.

감사합니다.

---
Hello, this is Hansol Kang from Rainbow Robotics.

Thank you for your valuable feedback on our RB-Y1.

Here is my response:
Currently, it is challenging to make port configurations variable within the existing structure. Additionally, the simulator operates normally when using port mapping instead of `network_mode=host`. As you may know, for X11 forwarding, the `xhost +` command needs to be executed in the terminal beforehand.

Thank you.

[image]
[image]

geniuskpj, 2024-11-08:
말씀하신대로 local에서는 동작하는데 ssh에서는 안되더라구요.
(option1은 ssh에서도 working)
local을 두명이 동시에 사용할 수는 없으니 자체적으로 해결방법 찾아보겠습니다.

closed 2024-11-08 by messy-snail