RainbowRobotics/rby1-sdk #41: Joint limitation for MJCF model

Skylark0924, 2024-12-27:
Thank you for the update of URDF model with accurate joint limits. 
I wonder when can we get the corresponding MJCF model with accurate parameters?

dongridong, 2025-01-03:
# Rainbow Robotics RBY1 MJCF Model

Thank you for your interest in our product!  
The latest MJCF model is now available [here](https://github.com/dongridong/mujoco_menagerie/tree/main/rainbow_robotics_rby1), and it will be updated in both the MuJoCo Menagerie and the RBY1-SDK main repositories soon.

If there are any further updates, we will let you know.  
Thank you for your continued support!

Skylark0924, 2025-01-05:
Thank you for replying. I found that position controllers are used in the current MJCF. Why not use joint torque limits?

dongridong, 2025-01-06:
Hello,

I have added content related to **joint torque limits** in the **actuator** section of the `rby1.xml` file on the `rainbow_robotics_rby1` branch.

Please refer to the following link:  
[MuJoCo Menagerie - rainbow_robotics_rby1.xml](https://github.com/dongridong/mujoco_menagerie/blob/rainbow_robotics_rby1/rainbow_robotics_rby1/rby1.xml)

Skylark0924, 2025-01-08:
Thanks a lot！！

closed 2025-01-08 by Skylark0924