RainbowRobotics/rby1-sdk #59: Gravity compensation controller does not exit after releasing button on RBY-1 wrist when servo is not enabled for all joints.

sageshoyu, 2025-01-15:
Steps to reproduce:

(1) Turn on servo for all joints in either the right _or_ left arm.
(2) Enable control.
(3) Press button on RBY-1 on servo/control-enabled arm and release.

The RBY-1 torso light remains green, and the control manager does not return to 'idle.'

When servo and control is enabled for all joints, this issue does not occur.

dongridong, 2025-01-15:
Hello, @sageshoyu 

We have reviewed the issue you inquired about.

The E-stop function has been confirmed to be operating normally.
Therefore, after engaging the E-stop, please perform a full servo-on to return the robot to the zero pose.

We will leave another comment once the bug has been fixed.

(If direct teaching for a single arm is required, please perform a servo-on for the torso (0–5) and the single arm (0–6), and then proceed with direct teaching.)

Thank you.

sageshoyu, 2025-01-15:
Yes, we do confirm the E-stop functions on our end. This triggers a Major Fault, which is expected. After power-cycling the R-PC to clear the major fault, we also confirm that we can servo-on all the joints to move the robot to a nominal position.

dongridong, 2025-01-17:
We have identified two bugs:

1. When only the right/left arm is in servo-on mode, attempting direct teaching does not disable gravity compensation mode.

2. When only the torso is in servo-on mode, the direct teaching button becomes disabled.

We plan to address these issues promptly and deploy a fix soon.

dongridong, 2025-01-17:
The bugs have been fixed.

You can find the updated version here: [RainbowRobotics v0.3.1 Release](https://github.com/RainbowRobotics/rby1/releases/tag/v0.3.1).

Please test it.

sageshoyu, 2025-01-22:
I updated to v0.3.2. The update appeared to be successful, except I received the following error:
```
Connection to 192.168.30.1 closed by remote host.
Error rebooting the server.
```

Regardless, when I servo on `left_arm0` through `left_arm_6` (and no other joints), and I am not able to enter and exit(!) gravity compensation mode by pressing the button on the RBY1 (left) wrist. 

I cannot servo on right_arm_6, but I think that is a separate issue (persisted after downgrading back to v0.3.0).

closed 2025-01-22 by sageshoyu