RainbowRobotics/rby1-sdk #69: Link number(id) for ComputeTransformation

anubhav-dogra, 2025-01-27:
Hi, 
Can you please provide an example on how to get/verify corresponding link name and id to be used in Compute Transformation. 
I am using the following functions:
```
robot = rb::Robot<rb::y1_model::A>::Create(address);
dyn = robot->GetDynamics();
        dyn_state = dyn->MakeState({"base"}, rb::y1_model::A::kRobotJointNames);

robot->StartStateUpdate(
        [this](const auto& state) {
            q_joint_wheels = state.position.block(0, 0, 2, 1);
            q_joint_waist = state.position.block(2, 0, 6, 1);
            q_joint_right_arm = state.position.block(2 + 6, 0, 7, 1);
            q_joint_left_arm = state.position.block(2 + 6 + 7, 0, 7, 1);
            q_joint_head = state.position.block(2 + 6 + 7 + 7, 0, 2, 1);
            
            q_set  << q_joint_waist, q_joint_waist, q_joint_right_arm, q_joint_left_arm, q_joint_head;
            dyn_state->SetQ(q_set);
            dyn->ComputeForwardKinematics(state_urdf);

            const auto& T_left = dyn->ComputeTransformation(dyn_state, 0, 4);
            const auto& T_right = dyn->ComputeTransformation(dyn_state, 0, 23);

            std::cout << "T_right: \n" << T_right << std::endl;
            std::cout << "T_left: \n" << T_left << std::endl;
            
            },1);
```

When I am printing link_names and joint_names, they are in very random order. 
Can you please help me in that? 
I think i am using the correct ids, with hit and trial. But please verify. 

Thanks.

messy-snail, 2025-02-03:
Sorry for the delayed response. We had a long national holiday in Korea.  

I am not sure what you are trying to achieve with indices 4 and 23.  
We assign indices based on the explicitly defined link names.  

For example, if you want to compute the transformation between the base and the right arm end-effector, you should define the state like this:  

```cpp
auto dyn_state = dyn->MakeState({"base", "ee_right"}, y1_model::A::kRobotJointNames);
```

Then, you can compute the transformation like this:  

```cpp
dyn->ComputeTransformation(dyn_state, 0, 1);
```

Let me know if you need further clarification.

messy-snail, 2025-02-03:
This topic has been further discussed here: https://github.com/RainbowRobotics/rby1-sdk/discussions/72.
Please refer to the link for more details and to continue the discussion.

closed 2025-02-03 by messy-snail

dongridong, 2025-02-03:
## Implementing Functionality in Python

If you want to implement functionality based on Python, please refer to the example below.

```python
dyn_state = dyn_robot.make_state(
    ["base", "link_right_arm_5", "link_left_arm_5", "link_torso_5", "ee_right", "ee_left"],
    Model_A().robot_joint_names
)
```

To compute the transformation matrix, use the following approach:
```python
right_T = dyn_robot.compute_transformation(dyn_state, 0, 4)   # base -> link_right_arm_5
left_T  = dyn_robot.compute_transformation(dyn_state, 0, 5)   # base -> link_left_arm_5
torso_T = dyn_robot.compute_transformation(dyn_state, 0, 3)   # base -> link_torso_5
```