RainbowRobotics/rby1-sdk #71: Real-time Control Example Not Working.

anubhav-dogra, 2025-01-31:
We @JafarAbdi tried to run real-time control example from the rby1-sdk, and the robot's controller manager goes to the major fault state. 
We always start it from home position, as coded in the example, it goes to the predefined state and when the control loop starts it gets stuck there after a few cycles (immediately), with controller major fault. 
Can you help us with this regards?
thanks.

messy-snail, 2025-02-03:
Sorry for the delayed response. We had a long national holiday in Korea.  

Our example is not well-optimized. Apologies for that. From the attached 
[video](https://rainbowco-my.sharepoint.com/:v:/g/personal/rby_support_rainbow-robotics_com/EWAmFhXSjnlFmOhLnW3oq7cBMQHqVNMfaA9tAhwXqeixwQ?nav=eyJyZWZlcnJhbEluZm8iOnsicmVmZXJyYWxBcHAiOiJPbmVEcml2ZUZvckJ1c2luZXNzIiwicmVmZXJyYWxBcHBQbGF0Zm9ybSI6IldlYiIsInJlZmVycmFsTW9kZSI6InZpZXciLCJyZWZlcnJhbFZpZXciOiJNeUZpbGVzTGlua0NvcHkifX0&e=1w4pgI), we can see that the robot moves extremely fast when executing real-time commands. This speed may exceed what our hardware can physically support, likely causing an error that triggers a fault state.  

You can check the **robot state table in the web UI** to see which indicator turns red and **check the log messages** to identify the specific issue.  

We recommend adjusting the real-time command parameters to ensure the robot moves within its supported limits. For example, you can modify the control input like this:  

```cpp
input.mode.setConstant(kPositionControlMode);
input.target = state.position * 0.99;
input.feedback_gain.setConstant(2);
input.feedforward_torque.setConstant(0);

input.finish = (count++) > 10000;
```

The recommended approach is to first validate the behavior in the [simulator](https://rainbowrobotics.github.io/rby1-dev/docs/simulators/) before running it on the actual hardware.  

If the issue persists, please provide details about **which indicator in the robot state table turns red and what messages appear in the logs** in the web UI. This information will help us analyze the problem further.  

Thanks.

dongridong, 2025-02-03:
## Defining Commands in Python

In Python, commands can be defined as follows.  
The first argument of `add_target` (**link**) is the reference value, and the second argument (**link**) is the target value.

```python
rc = RobotCommandBuilder().set_command(
    ComponentBasedCommandBuilder().set_body_command(
        BodyComponentBasedCommandBuilder()
        .set_torso_command(
            CartesianCommandBuilder()
            .add_target("base", "link_torso_5", T_torso, LINEAR_VELOCITY_LIMIT,
                        ANGULAR_VELOCITY_LIMIT, ACCELERATION_LIMIT / 2)
            .set_minimum_time(MINIMUM_TIME)
            .set_stop_orientation_tracking_error(STOP_ORIENTATION_TRACKING_ERROR)
            .set_stop_position_tracking_error(STOP_POSITION_TRACKING_ERROR)
        )
        .set_right_arm_command(
            CartesianCommandBuilder()
            .add_target("base", "ee_right", T_right, LINEAR_VELOCITY_LIMIT,
                        ANGULAR_VELOCITY_LIMIT, ACCELERATION_LIMIT / 2)
            .set_minimum_time(MINIMUM_TIME)
            .set_stop_orientation_tracking_error(STOP_ORIENTATION_TRACKING_ERROR)
            .set_stop_position_tracking_error(STOP_POSITION_TRACKING_ERROR)
        )
        .set_left_arm_command(
            CartesianCommandBuilder()
            .add_target("base", "ee_left", T_left, LINEAR_VELOCITY_LIMIT,
                        ANGULAR_VELOCITY_LIMIT, ACCELERATION_LIMIT / 2)
            .set_minimum_time(MINIMUM_TIME)
            .set_stop_orientation_tracking_error(STOP_ORIENTATION_TRACKING_ERROR)
            .set_stop_position_tracking_error(STOP_POSITION_TRACKING_ERROR)
        )
    )
)

anubhav-dogra, 2025-02-04:
Here's the table screenshot for your reference. 
We tried changing and adapting your recommendation but the behaviour is the same. 
When the real-time loops starts, robot immediately goes to Controller manager fault. 

[image]

messy-snail, 2025-02-20:
This issue has been discussed via email.
The response provided in the email is as follows:

---
regarding RT control:
As mentioned in the issue, if you send values exceeding the motor speed limits, the system will shut down. Since real-time control operates at **500Hz**, excessive angular changes within this short time may cause instability. The reference values should be given smoothly.

Keep in mind that **RT control means you are developing your own custom controller**. This is different from the default controllers we provide, such as **Joint Position, Joint Velocity, Cartesian control, etc**. If your goal is to control the robot using a predefined method, it may be better to use the built-in controllers rather than developing a new RT controller.

If you simply want to generate your own trajectory, **stream mode alone may be sufficient without RT control**. This allows you to send trajectory updates efficiently without implementing a fully real-time controller.

For more details on stream mode, please refer to:
-  [Robot and Command](https://rainbowrobotics.github.io/rby1-dev/docs/development/rby1-sdk/robot-and-command.html)
-  [Stream Mode](https://rainbowrobotics.github.io/rby1-dev/docs/development/rby1-sdk/controllers/control-hold-time-and-minimum-time.html#stream-mode)