RainbowRobotics/rby1-sdk #82: Servo failed on (Left_arm_6)

jiafei1224, 2025-02-22:
Hi, I encountered an issue with one of the components (left_arm_6). I've included some images for reference.

Note that left arm 6 has a red indication (the right arm is red because I deactivated it). Also, the area pictured became really hot during the process, but the temperature indication in the UI showed 30 degrees. The error said Actuactor Not Ready (Servo On Failed).

How can I fix this? Thanks.

[image]
[image]
[image]

messy-snail, 2025-02-24:
First, turn off all power and then turn on only the 48V power supply. At this point, rotate only the left arm’s Yaw2 joint and check if the encoder value changes.  

- **If the encoder value changes:** The motor itself is likely functioning properly. Proceed to the next step.  
- **If the encoder value does not change:** There could be a hardware issue, and you should contact rby.support@rainbow-robotics immediately.  

If the encoder value is changing normally, make sure the tool flange button is facing outward. This position corresponds to being near zero. **It’s important to note** that if the homing was previously set incorrectly, the zero value displayed in the UI might not be accurate. Don’t rely solely on the displayed value; instead, physically check if the tool flange is actually facing outward. For the left arm, it should look like the attached photo.  

[image]

After confirming the correct position, try performing the homing (Servo On) action. The issue might be that the homing position is continuously colliding with one of the limits.  

Also, note that the temperature displayed in the UI is an estimate and may not reflect the actual temperature. If you run into any issues while following these steps, please email rby.support@rainbow-robotics.

jiafei1224, 2025-02-24:
what is tool flange?

jiafei1224, 2025-02-24:
Furthermore, how do i rotate it (Left_arm_6) unless i release the break for that joint.

messy-snail, 2025-02-24:
The tool flange button is located at the end of the robot.
If you check the photo uploaded in this #81, you’ll see that the tool flange button is completely rotated in the opposite direction, which is not ideal.

Additionally, the Yaw2 joint does not have a separate brake. If the servo is on, it won't rotate. Please try turning off the power once and then rotate it.

If it still doesn’t rotate after turning off the power, please send an email to the address mentioned above. There is a high possibility of hardware failure.

helen9975, 2025-02-24:
Now we've rotated to the picture shown with the button facing outwards. The encoder values are changing. After servo on, the motor moves, but the error is still there.

messy-snail, 2025-02-24:
Could you show the error screen? Is it an init error or a big error?

jiafei1224, 2025-02-24:
Check out this video: https://drive.google.com/file/d/1QHyz-IPFDRakLL5AbMnYeJT-5LXDPqNC/view?usp=drivesdk

messy-snail, 2025-02-24:
I have reviewed the attached video carefully.
It doesn’t seem like the motor is burned. You mentioned earlier that the encoder values are changing—could you check if they are changing as expected?
I’m also curious if the values change normally in both the positive and negative directions.

jiafei1224, 2025-02-24:
https://us05web.zoom.us/j/83931361117?pwd=vOWZCQRnLbHJ4lwW3UTi4WO4DTPHbb.1

messy-snail, 2025-02-24:
That's a good idea.
However, we are currently in a morning meeting here, so it’s difficult to join right away.

messy-snail, 2025-02-24:
Sorry for the inconvenience, but could you send an email to rby.support@rainbow-robotics.com?
This issue seems to involve hardware-related problems, so it’s difficult for me to resolve it alone.
It would be best to schedule an official meeting with the hardware and firmware teams present.

jiafei1224, 2025-02-24:
Ok, i have emailed to schedule for something.

closed 2025-03-20 by jsoh91