RainbowRobotics/rby1-sdk #100: `version.h` not found when building SDK

sea-bass, 2025-04-12:
I was going through the ROS 2 repo setup and found an issue when building the `rby1-sdk`. Specifically:

```
--- stderr: rby1_hardware                                                                           
In file included from /home/sebastian/workspace/rby1_ws/src/rby1-ros2/rby1_hardware/include/rby1_hardware/rby1_hardware_interface.hpp:11,
                 from /home/sebastian/workspace/rby1_ws/src/rby1-ros2/rby1_hardware/src/rby1_hardware_interface.cpp:1:
/home/sebastian/workspace/rby1_ws/sdk/rby1-sdk/include/rby1-sdk/robot.h:3:10: fatal error: version.h: No such file or directory
    3 | #include "version.h"
      |          ^~~~~~~~~~~
```

I notice that `robot.h` does include this here: https://github.com/RainbowRobotics/rby1-sdk/blob/16cbbfdab5f6c5ea6897a8734741f7ccdeb9aaa0/include/rby1-sdk/robot.h#L3

However, there is no `version.h` available in the repo. Any ideas?

Since I have C++20 support (I'm on Ubuntu 24.04 / ROS 2 Jazzy), I could simply swap this out for the system header `#include <version>` and compilation works.

messy-snail, 2025-04-14:
Sorry for the late response.

This issue isn’t caused by the `<version>` system header — it’s actually related to the custom `version.h` we use.  

> [!WARNING]
> Please note that our SDK has only been tested on Ubuntu 22.04 with ROS2 Humble so far.

But the current build error seems to come from the SDK and is unrelated to the `version.h` issue.  
Please comment out the `#include "version.h"` line and try building again.  
This is definitely a bug on our side.

As shown here:  
https://github.com/RainbowRobotics/rby1-sdk/blob/16cbbfdab5f6c5ea6897a8734741f7ccdeb9aaa0/CMakeLists.txt#L73-L76
`version.h` is supposed to be generated during the CMake process, but it seems to not be working properly right now.  
So for now, **just remove or comment** out that line.

This functionality is only for version display and doesn’t affect robot operation at all.

sea-bass, 2025-04-15:
Thank you for the response, and generally for looking into this!

I am aware this is not supported on Jazzy, but I decided to try on my home PC to see how far I would get (I could get the MoveIt example working!). At work, where our robot will be, we are on Humble.