RainbowRobotics/rby1-sdk #110: RBY1 Firmware upgrade information? Can we update it ? how do we update it?

joestg, 2025-05-13:
Currently we just received our RBY1, It came with version 0.3.0 of the SDK. We upgraded our pip and sdk version to 0.6.0. 

We notice the new examples from the sdk 0.6.0 do not work on our robot.  (LED interface, firmware check)

The check firmware python example does not show any info for FW version.  

nvidia@tegra-ubuntu:~$ python ./rby1-sdk/examples/python/18_check_firmware_version.py --address 192.168.30.1:50051 
right_wheel, False, , 
left_wheel, False, , 
torso_0, False, , 
torso_1, False, , 

nvidia@tegra-ubuntu:~$ python ./rby1-sdk/examples/python/14_led.py --address 192.168.30.1:50051 
#1 Red 0.5s
Traceback (most recent call last):
  File "/home/nvidia/./rby1-sdk/examples/python/14_led.py", line 66, in <module>
    main(address=args.address, model=args.model)
  File "/home/nvidia/./rby1-sdk/examples/python/14_led.py", line 14, in main
    robot.set_led_color(
RuntimeError: gRPC call failed. Code: 12(UNIMPLEMENTED)

messy-snail, 2025-05-13:
Hi, I’m Hansol Kang. This is not an official answer.

If your RBY1 initially came with SDK version 0.3.0, you should be able to check the RPC version directly on the Web UI dashboard. Scroll down, and you should see the version information below the joint table. This should also display the RPC version, which is important to ensure it’s up to date.

If you haven’t updated the RPC, you probably need to do so.
You can find the latest RPC version and installation instructions at the following link:
[rby1-release](https://github.com/RainbowRobotics/rby1-release/releases)

I hope this helps.

joestg, 2025-05-13:
The Web UI just shows error for the Version. 
[image]

Thank you for the help, We are uploading the RPC update now. It looks like the RPC update is 750mb, seems like a full firmware update.

messy-snail, 2025-05-13:
Oh, sorry about that. I actually left the company, but I was the one who developed that version display feature. As far as I remember, there was a bug in RPC 0.3.x where it appeared incorrectly. I think it was fixed in version 0.4.0. You can probably find the record in the release page I mentioned.

If you update to the latest RPC, the version should be visible in the web UI, and you can also see it on the OLED screen on the side of the backpack.

joestg, 2025-05-13:
So we ran the updater from the settings menu using the newest 0.6.0 model A release. 
However this did not fix the issue and we are not sure if it updated. 

When we run robot 01_hello_rby1   we get an sdk commit from version 0.3.0

sdk_commit_id='e8db0b3'

We also tried the install instructions found here. 
https://rainbowrobotics.github.io/rby1-dev/docs/rpc-release/ 

This does not seem to update the SDK version on the RPC.

joestg, 2025-05-13:
Also the RPC version is not visible on the LCD. @RainbowRobotics

messy-snail, 2025-05-13:
The update function in the settings tab does not work.
Did you also try using the updater program from the release page?
You must wait until the system reboots properly.

It may be helpful to show the console log.
If everything works correctly, the system will reboot with a beep sound after the upload and extraction are completed.