RainbowRobotics/rby1-sdk #113: Fail to servo on

KkkyleZ, 2025-05-21:
Robot Right_Arm_6 has problem and I can not enable the control manager. When I tried to enable the control manager, it will cause major fault and turn off the power.

[image]

messy-snail, 2025-05-21:
First of all, this is not an official answer.

But based on common cases — the last joint (Right\_Arm\_6) rotates to find its home position. The fact that `init` failed on the dashboard likely means it failed during the homing process.
The Control Manager can only be enabled if all joints with Servo On commands are actually responding normally. If even one joint fails to turn on properly, control won't be allowed.

So, first check if the last joint (Right\_Arm\_6) rotates to find its home position. If it doesn't move or gets unusually hot, there may be a hardware issue.
In that case, it's best to contact **support ([rby.support@rainbow-robotics.com](mailto:rby.support@rainbow-robotics.com))** for further assistance.

KkkyleZ, 2025-05-21:
It has unusually hot. But the joint looks good , it is on the home position. I have already contact the support.

messy-snail, 2025-05-21:
If the joint is unusually hot, that’s not normal — it could lead to motor damage if left unchecked.

Also, how did you confirm it's at the home position?
Did you judge based on the position of the direct teaching button on the tool flange?
Try turning off all power and check if the motor can be rotated manually by hand.
If the motor is burnt out, it won’t move even by hand.

Also, when you turn on the servo, check if only the left joint moves while the right joint stays still.

KkkyleZ, 2025-05-21:
It cannot rotated manually by hand when I turn off all power, and only the left arm can move when I turn on the servo, the whole right arm can't move, but in the control panel it only shows that there is something wrong with Right_Arm_6.

messy-snail, 2025-05-21:
Sorry if there was any misunderstanding.
What I meant was specifically the wrist yaw 2 joint (Right_Arm_6) — try rotating that joint by hand with all power turned off.

When the servo is on, the joints are in position lock by default, so it's normal that they don't move.
That's why I asked you to test it with the power off.

If Right_Arm_6 doesn't rotate even when the power is off, there's a high chance the motor is burned out.

Also, you mentioned the whole right arm doesn't move — could you clarify what you mean?
If you're referring to it not moving during servo on, that's expected because of position lock.

KkkyleZ, 2025-05-21:
What I mean is that when servo starts up, the gripper will open and close. But only the left arm's gripper can open and close. And when I try to enable the control manager, the robot will immediately lose power and display a major fault. I will try to manually rotate the joint with all power turned off tomorrow. Thanks for your advice!

messy-snail, 2025-05-21:
Well… first, the gripper doesn’t use servo on/off — that concept doesn’t apply.
You probably ran a teleoperation example or something similar.
In the `rby1-sdk`, only the robot is controlled — the gripper is treated as an accessory and is controlled separately via Dynamixel code.

Based on what you described, it sounds like the Right\_Arm\_6 motor may be damaged.
And if the gripper on the right arm didn’t respond during the teleop example,
there’s also a chance that the tool flange port on the right arm is malfunctioning.

For an accurate diagnosis, support from the Rainbow Robotics team is probably necessary.

closed 2025-05-21 by KkkyleZ